ARM-software/CMSIS-DSP #197: Compilation warning with option -Wmissing-prototypes

ABOSTM, 2024-07-15:
Hello,

With CMSIS-DSP v1.15.0
With AC6 toolchain 6.19
With compiler option -Wmissing-prototypes

I got a lot of warnings (~20) about missing prototypes during my project compilation:
ex:
```
 C:/myproject/cmsis_pack_root/ARM/CMSIS-DSP/1.15.0/Source/FastMathFunctions/arm_vlog_f16.c:75:11: warning: no previous prototype for function 'logf16_scalar' [-Wmissing-prototypes]
 C:/myproject/cmsis_pack_root/ARM/CMSIS-DSP/1.15.0/Source/DistanceFunctions/arm_boolean_distance_template.h:442:6: warning: no previous prototype for function 'arm_boolean_distance_TT_TF_FT' [-Wmissing-prototypes]
...
```
 
Of course, removing the compiler option -Wmissing-prototypes, I don't have warning anymore.
But I would like to keep this option.
Also, with other toolchain IAR/EWARM, I got errors instead of warnings. This is fatal.

Is it possible to fix theses warnings ?

christophe0606, 2024-07-16:
@ABOSTM Hi, tell me if latest commit is solving the issue.
If you're using CMSIS Packs for building and since I have not yet released a new pack with this fix, you can get the temporary one generated with this  commit : https://github.com/ARM-software/CMSIS-DSP/actions/runs/9951995394

Just get the artifact and unzip it. It contains the CMSIS-Pack for this commit.

ABOSTM, 2024-07-16:
@christophe0606 ,
Thanks for your reactivity.
Yes, warnings about prototypes are fixed ... but ...

Here is the status of my test with your commit (1.15.1-dev64)
* AC6 compiler, everything is right, no error, no warning

* GCC compiler: 1 error
```
C:/myproject/cmsis_pack_root/ARM/CMSIS-DSP/1.15.1-dev64/Source/TransformFunctions/arm_bitreversal_f16.c:32:1: error: unknown type name 'float16_t'; did you mean 'float64_t'?
    32 | float16_t * pSrc,
       | ^~~~~~~~~
       | float64_t
```

* IAR/EWARM: 2 warning, and 1 error (those issues were already present before the fix):
```
        sh[0] = temp;
              ^
 "C:\myproject\cmsis_pack_root\ARM\CMSIS-DSP\1.15.1-dev64\Source\DistanceFunctions\arm_dtw_path_f32.c",151  Warning[Pa093]: implicit conversion from floating point to integer
         sh[1] = temp;
               ^
 "C:\myproject\cmsis_pack_root\ARM\CMSIS-DSP\1.15.1-dev64\Source\DistanceFunctions\arm_dtw_path_f32.c",155  Warning[Pa093]: implicit conversion from floating point to integer

               if (signbit(y))
                           ^
 "C:\myproject\cmsis_pack_root\ARM\CMSIS-DSP\1.15.1-dev64\Source\FastMathFunctions\arm_atan2_f16.c",138  Error[Pe2797]: "__fp16" not allowed here

```

christophe0606, 2024-07-16:
@ABOSTM For the `float16_t` error with gcc, can you try compiling with `-mfp16-format=ieee`.
But if you're not targeting a Cortex-M with hardware acceleration for `float16_t` then this datatype is not really useful and you can avoid building it with `-DDISABLEFLOAT16`

I look at the other errors now ...

christophe0606, 2024-07-16:
@ABOSTM For the `float16_t` issue there is indeed a mistake in my latest commit. So option `-mfp16-format=ieee` will not make a difference.

ABOSTM, 2024-07-16:
@christophe0606 I tested the -mfp16-format=ieee (before seeing your last post)
 It makes a difference:
At least GCC compilation is successful

Note I am using a STM32U545RE which includes a M33 with fpu

christophe0606, 2024-07-16:
@ABOSTM I have made a new commit with additional fixes that should solve remaining issues (but github tests have not finished running so you may want to wait until those tests have passed before testing).

ABOSTM, 2024-07-16:
@christophe0606
Everything is now well compiling: AC6, GCC and IAR
without error or warning.
I don't even need to enable -mfp16-format=ieee  or -DDISABLEFLOAT16

Thanks a lot.

I guess it will be part of next version 1.15.1
Any date for this official release ?

christophe0606, 2024-07-18:
@ABOSTM New CMSIS-DSP pack v1.16.0 has been released.

closed 2024-07-18 by christophe0606

ABOSTM, 2024-07-18:
Great Thanks