ARM-software/CMSIS-DSP #198: Weighted sum method appears to be doing a weighted average operation

EdanSneh, 2024-07-16:
I was looking at the code for weighted sum and it appears to be doing a weighted average operation. Seems to be dividing by the sum of the weights at the end.

Weighted average: https://www.cuemath.com/data/weighted-average/
Weighted sum: https://www.slideshare.net/slideshow/weighted-sum-method-an-introduction/157822124#2 

https://github.com/ARM-software/CMSIS-DSP/blob/main/Source/SupportFunctions/arm_weighted_sum_f16.c#L68-L110

Please double check my work.

christophe0606, 2024-07-17:
@EdanSneh It is indeed a weighted average. So the name of the function (and the documentation) are not correct.

christophe0606, 2024-07-17:
Corrected in latest commit. I renamed the function

closed 2024-07-23 by EdanSneh