ARM-software/CMSIS-DSP #200: no definition for "arm_cfft_init_1024_f32"

vipinthomas123, 2024-07-19:
Apologies for a simple question here, I've followed the usage instructions and explored all option to resolve the issue but with no luck.

We are are having trouble in compiling the arm_fft_bin_example.ewp. There is linker error for the arm_cfft_init_1024_f32 function.

I have added below library -
C:\Program Files\IAR Systems\Embedded Workbench 9.2\arm\CMSIS\Lib\IAR\iar_cortexM4lf_math.a

I have added below path for header files -
..\IAR-CMSIS-DSP\CMSIS_5\CMSIS\Core\Include
..\IAR-CMSIS-DSP\CMSIS-DSP\Include
..\IAR-CMSIS-DSP\CMSIS-DSP\Include\dsp

Development environment
[image]

**Error displayed -** 

_Error[Li005]: no definition for "arm_cfft_init_1024_f32" [referenced from ..\IAR-CMSIS-DSP\Examples\cortexM4lf\Obj\arm_fft_bin_example\arm_fft_bin_example_f32.o]_ 

Are we missing any additional settings.

**Reference -** https://github.com/ARM-software/CMSIS_5/issues/1670

Additionally can anyone please confirm example is for 1024 samples but testInput_f32_10khz (test input data) is of 2048, it is always expected the data samples to be double of processing samples??

christophe0606, 2024-07-19:
@vipinthomas123 Can you tell me which version of CMSIS-DSP is provided with your IAR environment ?

vipinthomas123, 2024-07-26:
[image]

I believe it's CMSIS 5.7

One update

If I use arm_cfft_init_f32(&varInstCfftF32,1024), I am able to compile without any error.

vipinthomas123, 2024-07-26:
@christophe0606 can you pls confirm this same CMSIS library can be used for Cortex R4? Or if there is different repo, pls share the link. Thank you

christophe0606, 2024-07-26:
@vipinthomas123 You're using an older version. So you can initialize with `arm_cfft_init_f32` but it may not be good for code size since it prevents the linker from deducing which FFT lengths you really use.

You may use the old way of initializing by settings the field of `arm_cfft_instance_f32` using structures from `CommonTables/arm_const_structs_f32.c`. For instance, `arm_cfft_sR_f32_len512` for length 512. In that case, you don't use `arm_cfft_init_f32`.

Note that this method of initializing will not work with Helium instruction set.

The library is never tested on Cortex-R because we focus on Cortex-M and have recently started to look at Cortex-A.

It should work on `Cortex-R4` if it is supported by CMSIS-Core. If it is not supported by CMSIS-Core, you won't be able to build and you should open a github issue on the CMSIS-6 repository to ask for R4 support.

closed 2024-08-08 by christophe0606