ARM-software/CMSIS-DSP #201: Typos from recent `ARM_DSP_ATTRIBUTE` additions

renesas-kyle-finch, 2024-07-25:
It appears there are some unintended insertions of the new `ARM_DSP_ATTRIBUTE` in front of some function parameters rather than only in front of functions.

When defining `ARM_DSP_ATTRIBUTE` to `__WEAK` for example, using IAR generates the following build error:
`Error[Pe1835]: attribute "weak" does not apply here`

The places I have found (so far) to have this issue are:
```
Source/TransformFunctions/arm_cftt_radix2_f16.c:arm_cfft_radix2_f16()
Source/TransformFunctions/arm_cftt_radix2_f32.c:arm_cfft_radix2_f32()
Source/SupportFunctions/arm_bitonic_sort_f32.c:arm_bitonic_sort_f32()
Source/FilteringFunctions/arm_fir_f32.c:arm_fir_f32()
```

Would it be possible for these to be fixed and a v1.16.2 be released?

christophe0606, 2024-07-26:
@renesas-kyle-finch Thanks for reporting this. I am testing with the __WEAK attribute but I am not testing with IAR. It looks like other compilers are not catching this issue.

christophe0606, 2024-07-26:
@renesas-kyle-finch I have just pushed a fix. Tell me if it is ok before I generate the new release.

christophe0606, 2024-07-29:
@renesas-kyle-finch New pack v1.16.2 has been uploaded. It should become visible in IDEs soon.

closed 2024-08-08 by christophe0606