ARM-software/CMSIS-DSP #202: Question regarding `arm_rfft_fast_f32`

mryndzionek, 2024-07-29:
I'm trying to understand how to use `arm_rfft_fast_f32` in a way 100% compatible with how `numpy.fft.rfft` works. `numpy.fft.rfft` takes in `N` real samples and outputs `N/2 + 1` complex samples (for even `N`). Now, this is not how `arm_rfft_fast_f32` works, right? Just looking at the input/output sizes. Also, internally `arm_rfft_fast_f32` seems to be using `arm_cfft_f32` directly on `arm_rfft_fast_f32` input buffer. This means that input to `arm_rfft_fast_f32` needs to be complex, right?
So, what is needed to achieve the exactly same effect as `numpy.fft.rfft` in CMSIS-DSP?

Isn't it the case that for a 512-point real FFT I need to on my own:
 - create a 1024 (complex) input buffer and populate it appropriately with my 512-size data
 - use `arm_rfft_fast_f32` in the 1024 variant (`arm_rfft_fast_init_1024_f32`)
 - only use the firs 257 (N/2 +1) of the resulting 512 complex samples
 
Help is appreciated. Would be good to expand documentation regarding this.

christophe0606, 2024-07-29:
@mryndzionek `arm_rfft_fast_f32` has a weird output data format due to legacy. The idea was to use an output buffer of same size as the input buffer. But a `N` real FFT gives `N/2+1` complex numbers so `N+2` float values instead of `N`. Except that the first complex and last complex are both real.

So, the last real is copied into the imaginary part of the first output sample.

It is explained here : https://arm-software.github.io/CMSIS-DSP/latest/group__RealFFT.html

You can also look at an example from the Python wrapper. The example is computing the rfft with scipy then making the copy of the last sample to be able to compare with CMSIS-DSP:

https://github.com/ARM-software/CMSIS-DSP/blob/main/PythonWrapper/examples/testrfft_all.py

I agree this behavior of `arm_rfft_fast_f32` is not pretty but we do not want to break backward compatibility.

mryndzionek, 2024-07-29:
Thanks for prompt reply!

closed 2024-07-29 by mryndzionek