ARM-software/CMSIS-DSP #204: Path to FVPs has changed in Keil MDK 5.40

RonanSynnottArm, 2024-08-13:
The path to the FVP executables has changed in MDK 5.40, which means that the debug settings in the example projects needs to be updated.

Suggest also adding something similar to the below in the readme(s).

```
Depending on the version of MDK you have installed, the path to FVP executables may need to be updated.

* MDK versions before 5.37:    `Keil_v5/ARM/FVP`
* MDK versions 5.38, 5.39:     `Keil_v5/ARM/VHT`
* MDK version 5.40 or later:   `Keil_v5/ARM/avh-fvp/bin/models`
```

christophe0606, 2024-09-02:
I have made fixes into branch https://github.com/ARM-software/CMSIS-DSP/tree/examples

Can you check if it is solving your issues ?

RonanSynnottArm, 2024-09-02:
Adding the REMarks in the run batch files probably suffices.

closed 2024-09-03 by christophe0606