ARM-software/CMSIS-DSP #205: Entry point not set in examples

RonanSynnottArm, 2024-08-13:
Have checked `dot-product` example, but likely applies to other examples.

In the linker options, the entry-point is not set, and so the default C-library entry (`__main`) is used in the ELF.

Loading into some debuggers (such as uVision) hide this issue, as the starting address is read from the Vector Table.

Other debuggers start from the entry point, meaning that init code is not executed, and the examples do not work correctly (usually manifesting a Hard Fault when the code reaches `__fp_init`).

Solution is to simply add the following to the linker options:
```
--entry=Reset_Handler
```

christophe0606, 2024-09-02:
I have made fixes into branch https://github.com/ARM-software/CMSIS-DSP/tree/examples

Can you check if it is solving your issues ?

RonanSynnottArm, 2024-09-02:
Confirmed fixed in dot_product example. Same fix should equally work well with all examples.

closed 2024-09-03 by christophe0606