ARM-software/CMSIS-DSP #216: arm_rfft_f32 result is different with python fft result.

netjoon, 2024-10-31:
When I ran the FFT with the attached data samples, the result was significantly different from the Python FFT result. I tested it on a Cortex-M3 using the arm_rfft_f32 function and obtained the attached result. Using the same input on a PC with Python, the results were vastly different. Is this expected?
[fft_input_signal.txt](https://github.com/user-attachments/files/17593782/fft_input_signal.txt)
[image]
[image]

atm-hash, 2024-11-01:
what is your signal？

netjoon, 2024-11-01:
It it voltage level.

atm-hash, 2024-11-01:
I look the text.I guess the problem is caused by data format. You can test use float  in python

netjoon, 2024-11-01:
Yes, the input data is of float type. In Python, there is no specific data type for this. I used the NumPy FFT module:
Python
fft_result = np.fft.fft(fft_input_signal)

The fft_input_signal is float type, and the results are also in float. I also tested with MATLAB’s FFT, and the results were the same as Python’s, unlike with arm_rfft_f32. The only difference is in precision: arm_rfft_f32 uses single precision, while Python and MATLAB use double precision. The difference between single and double precision is significant.

christophe0606, 2024-11-04:
@netjoon By Python FFT you mean a FFT from Numpy or SciPy or do you mean the FFT from the CMSIS-DSP Python wrapper ?

I assume you mean Numpy FFT.

The RFFT F32 has a weird output format that we need to keep for backward compatibility.
It is explained [here](https://arm-software.github.io/CMSIS-DSP/latest/group__RealFFT.html)

In short, the output of the RFFT with N real samples should have N/2+1 complex samples which is N+2 real values.

The original design of this RFFT decided to use a buffer of N samples for output.

They use the property that samples[0] have a zero imaginary part and samples[N/2+1] has also a zero imaginary part.

So the sample [N/2+1] is dropped. Its real part is written into imaginary part of sample[0] !

It may be the problem you have.

christophe0606, 2024-11-04:
@netjoon 

I have just tested with the CMSIS-DSP Python API and compared to SciPy. I have same results (with the modification of the output explained above).

```python
import numpy as np 
import cmsisdsp as dsp

import numpy as np
from scipy import signal
import matplotlib.pyplot as plt
import scipy.fft

from numpy.testing import assert_allclose

# copy / paste your 2038 real samples here
src=np.array([])

fftLen = 2048 

ref=scipy.fft.rfft(src)

# Convert to arrau of real to make it easier to compare
# with CMSIS-DSP output format
referenceFloat=np.zeros(fftLen)
referenceFloat[0::2] = np.real(ref)[:-1]
referenceFloat[1::2] = np.imag(ref)[:-1]

# Copy samples[n/2+1] real part into samples[0] imaginary part
referenceFloat[1] = np.real(ref[-1])

rfftf32=dsp.arm_rfft_fast_instance_f32()
status=dsp.arm_rfft_fast_init_f32(rfftf32,fftLen)
result = dsp.arm_rfft_fast_f32(rfftf32,src,0)

assert(len(src) == fftLen)
assert(len(result) == fftLen)

assert_allclose(referenceFloat,result,rtol=3e-6,atol=2e-5)

plt.plot(np.absolute(result))

plt.show()
plt.plot(np.absolute(referenceFloat))

plt.show()
```

netjoon, 2024-11-04:
Thank you for your update. I now understand the buffer structure and was able to achieve the same result using the CMSIS ARM FFT function with Python code.

closed 2024-11-04 by netjoon