ARM-software/CMSIS-DSP #217: conversion between incompatible vector types warnings

jeromecoutant, 2024-11-04:
Hi

Build with IAR toolchain:

      uint16x8_t    curExtremValVec = vdupq_n_s16(Q15_ABSMAX);
                                      ^
"/xxx/Source/StatisticsFunctions/arm_absmin_no_idx_q15.c",59  Warning[Pe1757]: conversion between incompatible vector types

      uint32x4_t    curExtremValVec = vdupq_n_s32(Q31_ABSMAX);
                                      ^
"/xxx/Source/StatisticsFunctions/arm_absmin_no_idx_q31.c",60  Warning[Pe1757]: conversion between incompatible vector types

      uint8x16_t     curExtremValVec = vdupq_n_s8(Q7_ABSMAX);
                                       ^
"/xxx/Source/StatisticsFunctions/arm_absmin_no_idx_q7.c",62  Warning[Pe1757]: conversion between incompatible vector types

Thx

christophe0606, 2024-11-04:
@jeromecoutant It looks like a bug. The `max` versions have been corrected to use unsigned intrinsics but the `min` is still using signed ones.

jeromecoutant, 2025-07-07:
Hi @christophe0606 
I am now using ARM.CMSIS-DSP.1.16.2
but I still get the same issue :-(

christophe0606, 2025-07-07:
@jeromecoutant I don't think it is corrected in 1.16.2 that one year old.
It is corrected in the HEAD. Do you still have the problem with the HEAD version ?

closed 2025-09-09 by christophe0606