ARM-software/CMSIS-DSP #218: how to get atan2_coefs_f32?

Quintin-Z, 2024-11-06:
Could you please help me with this? How are the coefficients of the arm_atan_limited_f32 function obtained? Are they derived from polynomial curve fitting or Taylor expansion? I used MATLAB to generate the curve over the interval [0, 1] and performed fitting, but I found that the obtained coefficients do not match those in the code.

christophe0606, 2024-11-07:
@Quintin-Z It is a polynomial produced using minimax approximation. The Remez algorithm is used.

You probably won't be able to exactly reproduce the coefficients but you should be able to produce a polynomial with same error bound on the interval [0,1]

Quintin-Z, 2024-11-07:
i got it，thank you very much！Mr. Favergeon

closed 2024-11-07 by Quintin-Z