ARM-software/CMSIS-DSP #220: Intrinsic definition collision for ARMv7-A/Cortex-A9

stellar-aria, 2024-12-13:
Compiler: GCC 13.2.1 (Arm GNU Toolchain)
Target: Cortex-A9 (`-mcpu=cortex-a9 -mfpu=neon`)
Flags: `-DNEON`

With the addition of #219, building now fails due to redefinition of certain intrinsics and macros previously defined at [CMSIS/Core/Incclude/cmsis_gcc.h:938](https://github.com/ARM-software/CMSIS_6/blob/4b7fcfa05d74c307ff10fb621af60c5ea060e38b/CMSIS/Core/Include/cmsis_gcc.h#L938). There's a note at https://github.com/ARM-software/CMSIS-DSP/blob/4404d3e65bd367754da83a8dd1f5669496677231/Include/arm_math_types.h#L115-L116 that seems relevant, but if that were the case there wouldn't be a name collision. 

Attempting to build _without_ providing an inclusion against `CMSIS/Core/Include` instead raises a different error: "fatal error: cmsis_compiler.h: No such file or directory".

I assume that Core must have its includes available for the DSP module, but doing so creates the collision. Deleting the check against `ARM_MATH_NEON` in `arm_math_types.h` removes the issue.

christophe0606, 2024-12-13:
@stellar-aria You can't build by just defining `ARM_MATH_NEON` and `__GNUC_PYTHON__` ?

`__GNUC_PYTHON__` will disable the inclusion of `cmsis_gcc.h` (and the dependency to CMSIS-Core) and `ARM_MATH_NEON` will enable the Neon variants when available.

christophe0606, 2024-12-13:
Above flags are for `make`. With `cmake` it should be `-DNEON=ON` and `-DHOST=ON`

christophe0606, 2024-12-13:
I have made a commit for this issue. It should solve the problem with CMSIS Core if you build with above defines.

stellar-aria, 2024-12-19:
Yes, removing the CMSIS Core reference and adding HOST ON worked, thanks so much!

closed 2024-12-19 by stellar-aria