ARM-software/CMSIS-DSP #228: Avoiding input modification in arm_rfft_fast_f32

Alex-EEE, 2025-01-07:
Hi,

I'm wondering if there is a way to avoid modifying the input in arm_rfft_fast_f32.  I see that the complex fft is in place using the input buffer.  I tried copying the input to pOut, run the cfff on pOut, then running the stage function with pOut as both input and output, but there's slight error introduced doing this.

```
      /* Calculation of RFFT of input */
      arm_cfft_f32( Sint, p, ifftFlag, 1);

      /*  Real FFT extraction */
      stage_rfft_f32(S, p, pOut);
   }
```

Is there any way to do the stage function in place?

christophe0606, 2025-01-07:
@Alex-EEE It is not possible to do the stage function in place since it is iterating the input buffer from beginning and from the end  in the same loop to compute the output.

I think the only way to avoid changing the input is by introducing a temporary buffer and making a copy.

Alex-EEE, 2025-01-07:
That's what I figured, but I thought I'd ask!  Getting to do an N pt FFT as N-1 means you still come out ahead vs in place complex FFT, and avoid the zero stuffing.  Typically I have to copy the input out of the original stream and zero pad it anyway, so it's usually not a loss to lose the input buffer anyway.

Thanks!

closed 2025-01-07 by Alex-EEE