ARM-software/CMSIS-DSP #231: When the "ARM_MATH_NEON" flag is set, the results from the "arm_biquad_cascade_df2T_f32" function do not match with the MATLAB simulation outcomes. However, the results from the optimized C code do match the simulation results.

mw-kakkala, 2025-01-16:
We are utilizing CMSIS-DSP version 1.14.3 to produce optimized code through the MATLAB workflow, targeting ARM Cortex-A family devices with CRL.
     We've enabled the 'ARM_MATH_NEON' flag to leverage the NEON SIMD instruction set. However, after generating the optimized code using this flag, the results do not align with the simulation outcomes. Conversely, when we use the default C implementation of the 'arm_biquad_cascade_df2T_f32()' function from CMSIS-DSP, the results match the simulation. Please let me know if there are any updates regarding this issue.

Regards,
Kishor

christophe0606, 2025-01-16:
@mw-kakkala Do you have a reproducer you can share ? Like some filter coefficients and a sample input ?

Note that in Neon mode, `arm_biquad_cascade_df2T_init_f32` do not take the filter coefficients directly. You need to use `arm_biquad_cascade_df2T_compute_coefs_f32` to generate a new array that can be passed to the init function

It is detailed in the documentation for `arm_biquad_cascade_df2T_init_f32`

christophe0606, 2025-01-23:
I see latest version of CMSIS-DSP have disabled the Neon version of this kernel. So, it looks like it indeed has issues.

rdguerrerom, 2026-09-12:
Some measurements on why the Neon version doesn't pay here, since it bears on whether this stays open.

**The structural reason.** I re-enabled the Neon path locally at `c0c8640` and timed it against the scalar one, single channel, ns per sample (Apple M5 / clang, minimum of 9 runs):

| numStages | blockSize | scalar | Neon | |
|--:|--:|--:|--:|--:|
| 2 | 128 | 3.58 | 6.19 | 0.58x |
| 2 | 512 | 4.40 | 7.16 | 0.61x |
| 4 | 128 | 5.53 | 7.16 | 0.77x |
| 4 | 512 | 7.98 | 7.57 | 1.05x |
| 8 | 128 | 10.09 | 13.02 | 0.77x |
| 8 | 512 | 14.89 | 14.00 | 1.06x |

So the assessment in #186 — *"the performance improvement is also not great"* — looks right, and I think it could not have been otherwise. A biquad is a recurrence in time: `y[n]` needs `y[n-1]`, and df2T carries `d1`/`d2` from one sample to the next. Within a single channel there is very little for SIMD to work on, so vectorising inside `arm_biquad_cascade_df2T_f32` is working against the dependency rather than with it. That is a property of the filter rather than of the implementation, so no amount of intrinsics in this function changes it.

**Where the parallelism actually is.** Across channels. An N-channel montage is N independent recurrences sharing one set of coefficients, and along that axis there is no dependency at all. It cannot be reached through this signature, which takes one signal and one state — so 64 channels is 64 calls, each running a serial recurrence.

With the channel index innermost, compared against the scalar version given its own preferred channel-major layout (so no de/interleave is charged to it), ns per sample per channel:

| channels | numStages | blockSize | current | channel-innermost | |
|--:|--:|--:|--:|--:|--:|
| 32 | 4 | 32 | 2.81 | 1.47 | 1.9x |
| 32 | 4 | 128 | 4.03 | 1.40 | 2.8x |
| 64 | 2 | 64 | 2.06 | 0.53 | 3.8x |
| 64 | 4 | 128 | 4.03 | 1.56 | 2.5x |
| 128 | 4 | 128 | 4.03 | 1.65 | 2.4x |

Portable C, no intrinsics. Agreement with the scalar output is about 4.5e-07, i.e. f32 rounding. State is the same 2 floats per stage per channel either way, so the footprint is identical. Below roughly 8 channels it loses, so this would be an additional entry point rather than a replacement — there is precedent in `arm_biquad_cascade_stereo_df2T_f32` for two channels.

**On the reported defect itself**, which may be useful independently of the above: with the Neon path re-enabled I could not get it to disagree with the scalar version at `c0c8640`. Agreement was 7e-08 to 2.7e-07 across numStages 1–8, blockSize 1–64, with coefficients passed through `arm_biquad_cascade_df2T_compute_coefs_f32` as documented, and with **distinct** per-stage coefficients — identical stages would mask an index error in the cascade-gain products that function precomputes. That is my own equivalence test rather than your test suite, so it is evidence that this may be stale rather than proof of it; it might be worth re-running the suite before anyone spends time on a fix.

Harnesses for all three measurements: https://github.com/rdguerrerom/cmsis-multichannel

Would a multi-channel entry point be of interest? Happy to put a patch together if so.