ARM-software/CMSIS-DSP #232: How to install the PythonWrapper on Raspberry PI after compiling the libCMSISDSP.a library?

PhilWareFenTech, 2025-01-21:
Hi,

Trying to install CMSISDSP with pip install fails on my Raspberry PI 3, so thought I'd compile it myself.
I've run CMake and make and it generated the bin_dsp/libCMSISDSP.a library.

How do I create the python module I can then install to my python3 system?

Thanks,
Phil.

christophe0606, 2025-01-22:
@PhilWareFenTech I hope at some point to find the time to deliver a binary version for Raspberry and that uses the Neon optimizations.

At the root (CMSIS-DSP folder where you can see the `setup.py` file), type:

`pip wheel . --no-deps -w dist`

You should now have a `dist` folder containing the package that you can install with a `pip install path_to_package`

PhilWareFenTech, 2025-01-22:
Thanks for your quick help. 

There seems to be a lot of things wrong though. 
I added the numpy version check with #prama message; 
PythonWrapper/cmsisdsp_pkg/src/cmsisdsp_module.h:41:9: note: ‘#pragma message: NPY_API_VERSION is 0x00000013’

So I'm using numpy-2.2

I've tried to pip install numpy-1.25, but it fails the same. 
Interestingly it complains the version is still 0x13 - which is numpy v2.2, but my Pip List says it's 1.25, and I verified it with checking the python envirnoment site-packages.

It feels like I've broken something...

```
(cmsis_env) phil@raspberrypi:~/CMSIS-DSP $ pip wheel . --no-deps -w dist
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Processing /home/phil/CMSIS-DSP
  Installing build dependencies ... done
  Getting requirements to build wheel ... done
  Preparing metadata (pyproject.toml) ... done
Building wheels for collected packages: cmsisdsp
  Building wheel for cmsisdsp (pyproject.toml) ... error
  error: subprocess-exited-with-error
  
  × Building wheel for cmsisdsp (pyproject.toml) did not run successfully.
  │ exit code: 1
  ╰─> [268 lines of output]
      running bdist_wheel
      running build
      running build_py
      copying cmsisdsp/version.py -> build/lib.linux-aarch64-cpython-311/cmsisdsp
      copying cmsisdsp/datatype.py -> build/lib.linux-aarch64-cpython-311/cmsisdsp
      copying cmsisdsp/fixedpoint.py -> build/lib.linux-aarch64-cpython-311/cmsisdsp
      copying cmsisdsp/mfcc.py -> build/lib.linux-aarch64-cpython-311/cmsisdsp
      copying cmsisdsp/__init__.py -> build/lib.linux-aarch64-cpython-311/cmsisdsp
      running egg_info
      writing cmsisdsp.egg-info/PKG-INFO
      writing dependency_links to cmsisdsp.egg-info/dependency_links.txt
      writing requirements to cmsisdsp.egg-info/requires.txt
      writing top-level names to cmsisdsp.egg-info/top_level.txt
      reading manifest file 'cmsisdsp.egg-info/SOURCES.txt'
      reading manifest template 'MANIFEST.in'
      adding license file 'LICENSE'
      writing manifest file 'cmsisdsp.egg-info/SOURCES.txt'
      running build_ext
      building 'cmsisdsp_transform' extension
      aarch64-linux-gnu-gcc -Wsign-compare -DNDEBUG -g -fwrapv -O2 -Wall -g -fstack-protector-strong -Wformat -Werror=format-security -g -fwrapv -O2 -fPIC -I/home/phil/CMSIS-DSP/Source/TransformFunctions -I/home/phil/CMSIS-DSP/Include -I/home/phil/CMSIS-DSP/PrivateInclude -IPythonWrapper/cmsisdsp_pkg/src -I/tmp/pip-build-env-qqop240n/overlay/lib/python3.11/site-packages/numpy/_core/include -I/home/phil/CMSIS-DSP/cmsis_env/include -I/usr/include/python3.11 -c PythonWrapper/cmsisdsp_pkg/src/cmsisdsp_transform.c -o build/temp.linux-aarch64-cpython-311/PythonWrapper/cmsisdsp_pkg/src/cmsisdsp_transform.o -Wno-attributes -Wno-unused-function -Wno-unused-variable -Wno-implicit-function-declaration -DCMSISDSP -D__GNUC_PYTHON__
      In file included from PythonWrapper/cmsisdsp_pkg/src/cmsisdsp_transform.c:32:
      PythonWrapper/cmsisdsp_pkg/src/cmsisdsp_module.h:41:9: note: ‘#pragma message: NPY_API_VERSION is 0x00000013’
         41 | #pragma message("NPY_API_VERSION is " STRINGIFY(NPY_API_VERSION))
            |         ^~~~~~~
      PythonWrapper/cmsisdsp_pkg/src/cmsisdsp_transform.c:1309:5: error: unknown type name ‘arm_dct4_instance_f32’
       1309 |     arm_dct4_instance_f32 *instance;
            |     ^~~~~~~~~~~~~~~~~~~~~
      PythonWrapper/cmsisdsp_pkg/src/cmsisdsp_transform.c: In function ‘arm_dct4_instance_f32_new’:
      PythonWrapper/cmsisdsp_pkg/src/cmsisdsp_transform.c:1339:46: error: ‘arm_dct4_instance_f32’ undeclared (first use in this function); did you mean ‘arm_dct4_instance_f32_new’?
       1339 |         self->instance = PyMem_Malloc(sizeof(arm_dct4_instance_f32));
            |                                              ^~~~~~~~~~~~~~~~~~~~~
            |                                              arm_dct4_instance_f32_new
      PythonWrapper/cmsisdsp_pkg/src/cmsisdsp_transform.c:1339:46: note: each undeclared identifier is reported only once for each function it appears in
      PythonWrapper/cmsisdsp_pkg/src/cmsisdsp_transform.c:1341:23: error: request for member ‘pTwiddle’ in something not a structure or union
       1341 |         self->instance->pTwiddle = NULL;
            |                       ^~
      PythonWrapper/cmsisdsp_pkg/src/cmsisdsp_transform.c:1342:23: error: request for member ‘pCosFactor’ in something not a structure or union
       1342 |         self->instance->pCosFactor = NULL;
            |                       ^~
      PythonWrapper/cmsisdsp_pkg/src/cmsisdsp_transform.c:1343:23: error: request for member ‘pRfft’ in something not a structure or union
       1343 |         self->instance->pRfft = NULL;
            |                       ^~
      PythonWrapper/cmsisdsp_pkg/src/cmsisdsp_transform.c:1344:23: error: request for member ‘pCfft’ in something not a structure or union
[... 207 lines cut ...]
      PythonWrapper/cmsisdsp_pkg/src/cmsisdsp_transform.c:1377:1: note: in expansion of macro ‘GETFIELD’
       1377 | GETFIELD(arm_dct4_instance_f32,normalize,"f");
            | ^~~~~~~~
      PythonWrapper/cmsisdsp_pkg/src/cmsisdsp_transform.c: In function ‘Method_arm_dct4_instance_f32_Nby2’:
      PythonWrapper/cmsisdsp_pkg/src/cmsisdsp_module.h:111:1: warning: control reaches end of non-void function [-Wreturn-type]
        111 | }
            | ^
[... cut ...]
            | ^
      PythonWrapper/cmsisdsp_pkg/src/cmsisdsp_transform.c:1375:1: note: in expansion of macro ‘GETFIELD’
       1375 | GETFIELD(arm_dct4_instance_f32,N,"h");
            | ^~~~~~~~
      error: command '/usr/bin/aarch64-linux-gnu-gcc' failed with exit code 1
      [end of output]
  
  note: This error originates from a subprocess, and is likely not a problem with pip.
  ERROR: Failed building wheel for cmsisdsp
Failed to build cmsisdsp
ERROR: Failed to build one or more wheels
```

christophe0606, 2025-01-22:
For NumPy "problem", I suggest you create a virtualenv. Install manually the v2.2 version. Build the package from this.

After, the resulting wheel should also work with older Numpy versions.

christophe0606, 2025-01-28:
With latest commit #234 it should now build (deprecated dct4 has been removed).
It is still pure C (no Neon optimization enabled in the wrapper).

closed 2025-02-25 by christophe0606