ARM-software/CMSIS-DSP #247: Memory Alignment Issues During arm_rfft_init_q31 Initialization

penglongzhao, 2025-03-31:
After enabling -flax-vector-conversions and -DARM_MATH_HELIUM flags:
When defining arm_rfft_instance_q31 rfft, the sizeof(rfft) returns 54 bytes, which is not aligned. However, during runtime, the memory access is performed with 4-byte alignment. This leads to illegal memory access (aligned access requires 60 bytes, but only 54 bytes were allocated during definition).
Question: How can this issue be resolved?
I've tried adding compiler options like -munaligned-access, -mno-unaligned-access, and -fpack-struct=4, but none of these solved the problem.
Adding #pragma pack(4) to the relevant structures in the CMSIS source code resolves the issue, but is modifying the source code the right approach? Are there better solutions?

christophe0606, 2025-03-31:
@penglongzhao Can you give me more details. What is the access to `arm_rfft_instance_q31` that is causing the illegal memory access ?

What is the compiler used and its version ?

penglongzhao, 2025-04-01:
## Environment
- CMSIS5 Version: commit 2b7495b85
- Repository: https://github.com/ARM-software/CMSIS_5.git
## Problem Description
By comparing the code between CMSIS5 and CMSIS-DSP repositories, it is found that the `arm_rfft_instance_q31` structure is defined identically in both repositories:

```c
typedef struct
{
    uint32_t fftLenReal;                        /**< length of the real FFT. */
    uint8_t ifftFlagR;                          /**< flag that selects forward (ifftFlagR=0) or inverse (ifftFlagR=1) transform. */
    uint8_t bitReverseFlagR;                    /**< flag that enables (bitReverseFlagR=1) or disables (bitReverseFlagR=0) bit reversal of output. */
    uint32_t twidCoefRModifier;                 /**< twiddle coefficient modifier that supports different size FFTs with the same twiddle factor table. */
    const q31_t *pTwiddleAReal;                 /**< points to the real twiddle factor table. */
    const q31_t *pTwiddleBReal;                 /**< points to the imag twiddle factor table. */
#if defined(ARM_MATH_MVEI) && !defined(ARM_MATH_AUTOVECTORIZE)
    arm_cfft_instance_q31 cfftInst;
#else
    const arm_cfft_instance_q31 *pCfft;         /**< points to the complex FFT instance. */
#endif
} arm_rfft_instance_q31;
```
## Potential Issues
When the compiler cannot guarantee global 4-byte alignment, using this structure may lead to memory access errors.

christophe0606, 2025-04-01:
@penglongzhao The compiler should be managing the padding so that there are no issues.
I have run the RFFT Q31 test suite and added:

```
#include <stddef.h>
_Static_assert(offsetof(arm_rfft_instance_q31, cfftInst) % _Alignof(arm_cfft_instance_q31) == 0, "cfftInst is not properly aligned!");
```

in the `dsp/transform_functions.h` header

I have tested with Arm Compiler 6.22, Clang 18.1.3 and gcc 13.3.1

In the 3 cases, there are no alignment issues. `cfftInst` is the only field accessed with `&` during initialization. Other fields are accessed normally so should not cause any problems.

Which compiler are you using ? What are the compilation flags ?

penglongzhao, 2025-04-01:
**Build Environment Note**  
Sorry, I forgot to mention the compiler earlier - I'm using the `arm-zephyr-eabi` cross-compiler from the `zephyr-sdk-0.17.0` toolchain.

**Alternative concise version:**  
`Compiler: arm-zephyr-eabi (zephyr-sdk-0.17.0)`

penglongzhao, 2025-04-01:
**Environment Observation**  
The application exhibits different behavior between toolchains:  

| Toolchain            | Compilation | Runtime Behavior |
|----------------------|-------------|------------------|
| `gcc` (version x.x)  | ✅ Success  | ✅ Works as expected |
| `arm-zephyr-eabi` (zephyr-sdk-0.17.0) | ✅ Success | ❌ Unexpected behavior |

penglongzhao, 2025-04-01:
### Root Cause Analysis  
The runtime behavior varies across compilers due to **alignment/padding differences**.  

#### Current Observations:  
- ✅ **Most compilers**: Work correctly (e.g., GCC, Clang)  
- ❌ **Specific toolchains**: Fail at runtime (e.g., `arm-zephyr-eabi` from zephyr-sdk-0.17.0)  
- 🔄 **Solution**: Explicitly enforcing 4-byte alignment via `#pragma pack(4)` resolves the issue universally  

#### Trade-offs:  
```c
#pragma pack(4)  // Requires modifying CMSIS headers

christophe0606, 2025-04-02:
@penglongzhao My understanding of the C specification is that if `S` is a lvalue then `S.xxx` is also a lvalue and the compiler will generate whatever padding is required by the architecture to enable this.

So it is valid to take the address `&S.xxx`. The CMSIS-DSP code is just relying on this. It is not relying on any specific padding values or even the existence of any padding.

As consequence, I think the C code is valid and I won't change it nor add any workaround.

If in the future if I am explained that the C code is not valid from the C standard point of view, then I'll prefer correct the C code than adding workaround with pragmas.

But for now, I'll keep the code as it is since it looks like a compiler related problem.

penglongzhao, 2025-04-07:
ok,thanks

closed 2025-04-07 by penglongzhao