ARM-software/CMSIS-DSP #251: Undefined reference to `logf'

PidsoEST, 2025-04-25:
I have cross-compiled the CMSIS-DSP into an archive and then I link it into it in my code. I'm using the CortexA9, thus I include the `CMSIS_CORE_INCLUDES` := $(CMSIS_5)/CMSIS/Core_A/Include` in the makefile.
Most CMSIS-DSP functions work fine, but when calling some functions like `arm_vexp_f32()` and `arm_sin_f32` I get a bunch of the following errors during linking:

```
ld.exe: FastMathFunctions.c:(.text+0x67c): undefined reference to `logf'
...
ld.exe: C:\Users\Fakename\MyStuff\LIBs\CMSIS-DSP\builddir\libCMSISDSP.a(FastMathFunctions.o):FastMathFunctions.c:(.text+0x6ac): more undefined references to `logf' follow
```
Does anybody know what I need to do to correctly link the `logf()` function?
I attached my makefile for reference.

Thanks

[Makefile.txt](https://github.com/user-attachments/files/19911716/Makefile.txt)

christophe0606, 2025-04-28:
@PidsoEST Are you linking at the end with `-lm`
Functions like `logf` are provided by the C compiler and its standard libraries. They are not part of CMSIS-DSP.

PidsoEST, 2025-04-30:
@christophe0606 Thanks for the hint, though I've tried that already and it didn't change anything unfortunately. The linker keeps failing with the same errors: `undefined reference to logf`, `undefined reference to expf`, ...

Is this the correct place to add the math lib inside the Makefile.txt?
```
# Compilation flags (here for Cortex-M55)
#--target=arm-none-eabi 
CFLAGS := -mcpu=cortex-A9 \
 -Wsign-compare \
 -Wdouble-promotion \
 -Ofast -ffast-math \
 -DNDEBUG \
 -Wall -Wextra  -Werror \
 -fshort-enums -fshort-wchar \
 -mfloat-abi=hard \
 -lm                            # <------- added here
```
Is there anything else I can try?

PidsoEST, 2025-04-30:
@christophe0606  You mean to the linker command inside my application rather than the cmsis-dsp makefile, correct?
There I'm already linking against math lib as well. I'll attach the linker invocation command (some .o files removed to not spam the chat):
```
Invoking: ARM v7 gcc linker
arm-none-eabi-gcc
-L"T:\sw_projects\sdk_workspace\trenz_fsbl_bsp\ps7_cortexa9_0\lib"
-L"C:\Users\Fakename\Mystuff\LIBs\CMSIS-DSP\builddir"
-mcpu=cortex-a9
-mfpu=vfpv3
-mfloat-abi=hard
-Wl,-build-id=none
-specs=Xilinx.spec
-Wl,-T
-Wl,../src/lscript.ld
-LE:/Projects/PID-410/Prj_repo/PID-410/SOW/trunk/PID-410-A.1CA-030_CPP/Rev_C/SW/Prj_VivadoPrj/Prj_firmware/Prj_firmware.sdk/trenz_fsbl_bsp/ps7_cortexa9_0/lib
-o
"Prj_RTOS.elf"

./src/qspi/xqspips.o
./src/qspi/xqspips_g.o
./src/qspi/xqspips_hw.o
./src/qspi/xqspips_options.o
./src/qspi/xqspips_selftest.o
./src/qspi/xqspips_sinit.o

./src/Utilities/AccessConfigInfo.o
./src/Utilities/Base64.o
./src/Utilities/Crc8.o
./src/Utilities/MathUtil.o
./src/Utilities/SystemUtils.o
./src/Utilities/TrackingFuncs.o
./src/Utilities/fast_math_neon.o
./src/Utilities/jsmn.o
./src/Utilities/ring_buffer.o

[...]

./src/FreeRTOS_asm_vectors.o
./src/FreeRTOS_tick_config.o
./src/flash_lqspi.o
./src/hrf_console_callbacks.o
./src/main.o
./src/platform.o

-lm
-lCMSISDSP
-Wl,--start-group,-lxil,-lgcc,-lc,--end-group
-Wl,--start-group,-lrsa,-lxil,-lgcc,-lc,--end-group
-Wl,--start-group,-lxilffs,-lxil,-lgcc,-lc,--end-group

```
Do you have an idea what else I could try to get it to link properly? 
Thanks for your time.

christophe0606, 2025-04-30:
@PidsoEST Try to put `-lm` at the end after the `-lCMSISDSP`.

Since the issue is Makefile and tool related and not CMSIS-DSP specific (you would have same issues in any code using `logf` or other functions defined in `math.h`), I consider this github issue as DONE.

PidsoEST, 2025-04-30:
@christophe0606 Changing the linking order fixed the problem, thanks a lot.

closed 2025-04-30 by PidsoEST