ARM-software/CMSIS-DSP #253: Testing/processResult.py and missing currentConfig.csv?

ilyesgouta, 2025-05-07:
Hi,

How to create currentConfig.csv?

The documentation tells that results could be extracted from the console output using the processResult.py script and once the CMSIS-DSP benchmark is run on the target. However the latter requires the file currentConfig.csv to be already present in the filesystem (same folder) otherwise it bails out with an error. However there seem to be no CMakeLists.txt in the Testing folder. I'm using CMSIS v5.9.0 and DSP v1.10.0.

Regards,
Ilyes

christophe0606, 2025-05-07:
@ilyesgouta Hi, you're right. The code for generating this has been removed from the `CMakeFile.txt` because it was not generic enough.

You can generate the `.csv` manually. It just  contains some informations to know the benchmark condition (defines for CMSIS-DSP, compiler version ...)

Here is the code that was used in the `CMakeFiles.txt`. I'll update the documentation to describe the format of this `.csv`

It is a two line `.csv`. A header and some data.

```cmake
function(writeConfig path)
  set(output "")
  list(APPEND output "OPTIMIZED,HARDFP,FASTMATH,NEON,HELIUM,UNROLL,ROUNDING,PLATFORM,CORE,COMPILER,VERSION\n")

  if (OPTIMIZED)
    list(APPEND output "1")
  else()
    list(APPEND output "0")
  endif()

  if (HARDFP)
    list(APPEND output ",1")
  else()
    list(APPEND output ",0")
  endif()

  if (FASTMATHCOMPUTATIONS)
    list(APPEND output ",1")
  else()
    list(APPEND output ",0")
  endif()

  if (NEON OR NEONEXPERIMENTAL)
    list(APPEND output ",1")
  else()
    list(APPEND output ",0")
  endif()

  if (HELIUM OR MVEI OR MVEF)
    list(APPEND output ",1")
  else()
    list(APPEND output ",0")
  endif()

  if (LOOPUNROLL)
    list(APPEND output ",1")
  else()
    list(APPEND output ",0")
  endif()

  if (ROUNDING)
    list(APPEND output ",1")
  else()
    list(APPEND output ",0")
  endif()

  list(APPEND output ",${PLATFORMID}")
  if (CONFIGID)
     # Specific config ID (like M55 with autovectorizer)
     list(APPEND output ",${CONFIGID},")
  else()
     # Core ID is used as config ID when not specified
     list(APPEND output ",${COREID},")
  endif()
  if (ARMAC6)
    list(APPEND output "AC6")
  elseif(GCC)
    list(APPEND output "GCC")
  endif()
  compilerVersion()
  list(APPEND output ",${COMPILERVERSION}")

  file(WRITE ${path} ${output})
  
 
endfunction()
```

ilyesgouta, 2025-05-09:
Hi @christophe0606 Thanks for the information. I believe it would be best to revert the deletion patch and restore that writeConfig function. It is better to have a functional build system than incomplete / broken.