ARM-software/CMSIS-DSP #261: arm_fir_interpolate_f32: out of bounds read of pState leading to bad output

amurzeau, 2025-08-03:
Hi,

I'm using arm_fir_interpolate_f32 with these parameters:
- 348 taps coefs
- L = 6
- `phaseLength` = 58 (348 / 6)
- `blockSize` = 8
- `pState` buffer of 65 floats (8 + 58 - 1 == 65)

According to https://arm-software.github.io/CMSIS-DSP/v1.15.0/group__FIR__Interpolate.html, pState should need only `blockSize + phaseLength - 1` floats.

But I get unexpected behavior, where when all input samples are 0 I still get some random outputs.
This is due to out of bounds read on pState array:
https://github.com/ARM-software/CMSIS-DSP/blob/70e56d0bcb21a6ceff254f84727c2a9b8ea587c1/Source/FilteringFunctions/arm_fir_interpolate_f32.c#L356-L383

As phaseLen == 58, the code enter the if for the last 2 taps (`tapCnt == 2`).
But when processing the last sample of the block (which is the 8th sample), pState is at offset 7 due to this line:
https://github.com/ARM-software/CMSIS-DSP/blob/70e56d0bcb21a6ceff254f84727c2a9b8ea587c1/Source/FilteringFunctions/arm_fir_interpolate_f32.c#L437

Then, inside the `if (tapCnt > 0U)`, there is a unconditional read of 4 floats instead of only `tapCnt` (2 in my case) from pState, via ptr1, which was at offset `7 + 56` before entering the `if`.
Reading 4 floats from there will reads floats at offset 63, 64, 65 and 66.
But offsets 65 and 66 are out of bounds of the pState array of `8 + 58 - 1 == 65` floats.

I've tried to replace the whole `if` block with this, and it seems to work correctly, but I've not done extensive tests to check the output:
```c
while (tapCnt > 0U)
{
  f32x4_t vecCoef;

  state0 = *ptr1;
  ptr1++;

  vecCoef = vldrwq_gather_shifted_offset_f32(ptr2, vec_stridesM);
  ptr2 += S->L;
  acc = vfmaq_n_f32(acc, vecCoef, state0);

  tapCnt--;
}
```

christophe0606, 2025-09-10:
The code use `vecState = vldrwq_z_f32(ptr1, p0);` which is using `ptr1`.

Anyway, with this most recent version I can't reproduce your problem. 
Can you test with the most recent version ?

amurzeau, 2025-09-10:
I can reproduce the issue with the latest commit on main (0d43ab11fb7f348ea04c9e9eba48fc7e0c73671c).
I'm using this test:
```cpp
#include "dsp/filtering_functions.h"
#include <assert.h>

void test_fir_interpolate_f32_test_out_of_bounds() {
    // Use FIR interpolate to "up sample" a silence by a factor of 6 using a all-1.0f taps coefs.
    // This should gives a silence in pDst.
 
    arm_fir_interpolate_instance_f32 SI;
    const int q = 6;
    const int numTaps = 348;
    const int blocksize = 8;
    const int dstsize = blocksize * q;
 
    float32_t* pSrc = (float32_t*) malloc(blocksize * sizeof(float));
    float32_t* pDst = (float32_t*) malloc(dstsize * sizeof(float));
 
    // Allocate 2 * 6 values more to test out of bounds access
    float32_t* pCoefs = (float32_t*) malloc((numTaps + 2 * 6) * sizeof(float));
 
    const int pStateSize = blocksize + numTaps / q - 1;
    // Allocate 4 values more to test out of bounds access
    float32_t* pState = (float32_t*) malloc((pStateSize + 4) * sizeof(float));
 
    memset(pSrc, 0, blocksize * sizeof(float));
    memset(pDst, 0, dstsize * sizeof(float));
 
    // All-1 coefs
    // Also set values after numTaps for out of bounds testing
    for(int i = 0; i < (numTaps + 2 * 6); i++) {
        pCoefs[i] = 1.0;
    }
 
    memset(pState, 0, pStateSize * sizeof(float));
 
    // Put non-0 values after pState array to test out of bounds access
    // If these values are used, they will be multiplicated with pCoefs (which are 1.0) and will cause
    // non-0 values in pDst output.
    pState[pStateSize + 0] = 1234.0;
    pState[pStateSize + 1] = 5678.0;
    pState[pStateSize + 2] = 8912.0;
    pState[pStateSize + 3] = 3456.0;
 
    arm_status status = arm_fir_interpolate_init_f32(&SI, q, numTaps, pCoefs, pState, blocksize);
    assert(status == ARM_MATH_SUCCESS);
 
    arm_fir_interpolate_f32(&SI, pSrc, pDst, blocksize);
 
    // All pSrc and pState values were 0, so we should only have 0 in pDst
    for(int i = 0; i < dstsize; i++) {
        assert(pDst[i] == 0.0f);
    }
 
    free(pSrc);
    free(pDst);
    free(pCoefs);
    free(pState);
}
```